kasbert/epsolar-tracer #2: Real Time Clock

tomturner123, 2015-10-08:
Hi, firstly thank you so much for putting this on github. I have been having a play with my tracer controller and it works great! One issue I have is that I would like the "load" to turn on and off at a certain time. However I assume this will need adjusting for daylight savings so I have been trying to do this remotely / automatically. I can successfully read the real time clock 1,2 & 3 but I cant for the life of me figure out how they represent the time / date. The description says seconds / minutes, hour / day etc but I dont really understand the results. e.g rtc1=6441 rtc2=2062 rtc3=3850 should be approx 2:30pm 8/10/15. Any light you could shed would be very gratefully received.

Additionally my issue is that I would need to change the time back one hour, I can successfully use the "write_output" command for other registers but it doesnt seem to have any effect when I use it on the rtc. The write_output command doesnt fail - no errors, it just doesnt change the result. I assume that this is because all the rtc commands must be changed simultaneously but its not very clear how to do that as write_output will only accept one argument.

Many thanks for any help you can give.

kasbert, 2015-10-09:
Hi, nice to hear that you find the code useful. Unfortunately my controller is about 500km away and I don't have remote access to it.

I think you are right, that the registers should be written simultaneously, but the write_output can handle only one register at a time. So one should either add new methods for writing and reading multiple registers at a time or enhance the current ones. The modbus client can write multiple subsequent registers.

dmdelorme, 2017-12-23:
i just worked though part of the problem.. it is a little hacky.
``#####read clock
result = client.read_holding_registers(0x9013,3,unit=1)
secmin = result.registers[0]
secs = (secmin & 0xff)
minuits = secmin >> 8
hrday = result.registers[1]
hr = (hrday & 0xff)
day = hrday >> 8
monthyear = result.registers[2]
month = (monthyear & 0xff)
year = monthyear >> 8
Still working on syncing time with computer.

pelom, 2018-02-14:
File: client.py
Class: EPsolarTracerClient

```
def readRTC(self):
        register = registerByName('Real time clock 1')
        sizeAddress = 3
        result = self.client.read_holding_registers(register.address, sizeAddress, unit=self.unit)
        return self.decodeRTC(result.registers)

    def writeRTC(self, datetime):
        register = registerByName('Real time clock 1')
        values = self.encodeRTC(datetime)
        self.client.write_registers(register.address, values, unit=self.unit)
        return True

    def decodeRTC(self, rtc):
        s = 2000
        secMin  = rtc[0]
        hourDay = rtc[1]
        monthYear = rtc[2]
        secs  = (secMin & 0xff)
        hour  = (hourDay & 0xff)
        month = (monthYear & 0xff)
        minut = secMin    >> 8
        day   = hourDay   >> 8
        year  = monthYear >> 8
        return datetime(s+year, month, day, hour, minut, secs)

    def encodeRTC(self, datetime):
        s = 2000
    	rtc1 = int( (datetime.minute  << 8) | datetime.second)
    	rtc2 = int( (datetime.day 	  << 8) | datetime.hour)
    	rtc3 = int( (datetime.year -s << 8) | datetime.month)
        return [rtc1, rtc2, rtc3]
```

kasbert, 2023-04-29:
RTC functions added, thanks @pelom

closed 2023-04-29 by kasbert