kasbert/epsolar-tracer #4: Error on Raspberry Pi, Kernel 4.1.12-v7+

frolov-alexey, 2015-11-02:
Steps to reproduce: 
- remove cdc_acm driver
- install xr_usb_serial_common driver
- execute: setserial -a /dev/ttyXRUSB0

and we can see an error `xr_usb_serial_ctrl_irq - urb shutting down with status: -2` in kernel log
```
Nov  2 20:25:28 raspberrypi kernel: [ 3195.724863] tty ttyXRUSB0: xr_usb_serial_tty_install
Nov  2 20:25:28 raspberrypi kernel: [ 3195.724939] tty ttyXRUSB0: xr_usb_serial_tty_open
Nov  2 20:25:28 raspberrypi kernel: [ 3195.724961] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_port_activate
Nov  2 20:25:28 raspberrypi kernel: [ 3195.725344] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_ctrl_msg - rq 0x22, val 0x3, len 0x0, result 0
Nov  2 20:25:28 raspberrypi kernel: [ 3195.727718] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_tty_close
Nov  2 20:25:28 raspberrypi kernel: [ 3195.727776] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_port_shutdown
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728060] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_ctrl_msg - rq 0x22, val 0x0, len 0x0, result 0
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728092] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_ctrl_irq - urb shutting down with status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728125] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728148] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728171] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728194] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728216] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728238] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728261] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728283] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728305] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728328] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728350] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728396] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728421] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728443] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728466] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728488] cdc_xr_usb_serial 1-1.5:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Nov  2 20:25:28 raspberrypi kernel: [ 3195.728574] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_tty_cleanup
```

frolov-alexey, 2015-11-02:
OK, actually in this fork there is no such error.

closed 2015-11-02 by frolov-alexey

Kutashov, 2017-03-06:
I have the same issue, regardless I used this fork or original xr driver. 
Have you fixed this somehow?

Lnce, 2017-03-30:
Now this error has become to my raspbian too, any ideas Kutashov? I can still randomly access data, but I'm afraid of lockups.

Kutashov, 2017-03-30:
Unfortunately, no.

Srendi, 2017-04-27:
metoo:
pi@solar:~/epsolar-tracer.GAV $ uname -r
4.9.24-v7+
pi@solar:~/epsolar-tracer.GAV $ sudo setserial -a /dev/ttyXRUSB0
/dev/ttyXRUSB0, Line 0, UART: unknown, Port: 0x0000, IRQ: 0
        Baud_base: 115200, close_delay: 5, divisor: 0
        closing_wait: 300
        Flags: spd_normal low_latency

pi@solar:~/epsolar-tracer.GAV $

cat /var/log/syslog:

Apr 27 10:21:26 solar kernel: [10234.735597] tty ttyXRUSB0: xr_usb_serial_tty_install
Apr 27 10:21:26 solar kernel: [10234.735639] tty ttyXRUSB0: xr_usb_serial_tty_open
Apr 27 10:21:26 solar kernel: [10234.735650] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_port_activate
Apr 27 10:21:26 solar kernel: [10234.736071] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_ctrl_msg - rq 0x22, val 0x3, len 0x0, result 0
Apr 27 10:21:26 solar kernel: [10234.736861] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_tty_close
Apr 27 10:21:26 solar kernel: [10234.736881] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_port_shutdown
Apr 27 10:21:26 solar kernel: [10234.737220] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_ctrl_msg - rq 0x22, val 0x0, len 0x0, result 0
Apr 27 10:21:26 solar kernel: [10234.737240] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_ctrl_irq - urb shutting down with status: -2
Apr 27 10:21:26 solar kernel: [10234.737262] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737280] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737297] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737313] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737329] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737345] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737366] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737382] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737402] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737418] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737434] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737449] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737465] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737480] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737496] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737511] cdc_xr_usb_serial 1-1.4:1.1: xr_usb_serial_read_bulk_callback - non-zero urb status: -2
Apr 27 10:21:26 solar kernel: [10234.737576] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_tty_cleanup

Lnce, 2017-04-27:
Beware of this driver, it works okay when reading data with those errors, but my raspberry died and rebooted randomly.
- kernel: [***] usb 1-1-port2: over-current change

That's why I won't risk it and have unloaded the module, if not reading data.