kasbert/epsolar-tracer #5: Illegal Address problem

csabyka, 2015-12-29:
Dear kasbert

I try to use your tracer, but i receive this error: 
python readall.py
{0: 'EPsolar Tech co., Ltd', 1: 'Tracer1210A', 2: 'V01.10+V02.11'}

<pyepsolartracer.registers.Register instance at 0x767cbaf8>
read_input_registers: 10000
read_holding_registers: Exception Response(131, 3, IllegalAddress)

Do u have any idea about this ? 

best regards 
csaba

kasbert, 2016-01-14:
Sorry for the delay. There no problem with that output. readall.py uses wrong methods on purpose. For example 0x3000:"Charging equipment rated input voltage" should be read with read_input_registers and not with read_holding_registers. readall.py uses both methods just to see what the controller answers. I used debug output to create testdata.py

closed 2016-01-14 by kasbert

csabyka, 2016-01-14:
Dear Kasbert. 
Can we talk in outside from github ? 
Please drop me an email, my address is csy blablablaatblabla csy dot hu .
Thank you. 
cs