kasbert/epsolar-tracer #9: Also supports A-Series (Tracer 4210A)

danieloneill, 2016-07-20:
Just a note, your excellent work is also compatible with at least the 4210A controller:

http://www.aliexpress.com/item/4210A-MPPT-40A-Solar-Charger-Controller-LCD-12V-24V-Auto-EPEVER-High-Efficiency-Regulador-Solar-with/32628136007.html

readall.py reports:

{0: 'EPsolar Tech co., Ltd', 1: 'Tracer4210A', 2: 'V01.12+V02.11'}

Though not entirely unexpected, I appreciate your work on this project and would like to provide some formal verification for A-series support.  If you have any additional tests which would help with development please don't hesitate to ask. So far, the BN-series registers match the A-series perfectly, though.

Lnce, 2016-08-22:
What cable do you use? Does the original "EPEVER" cable work? What kernel version?

danieloneill, 2016-08-31:
Yep, original cable. I did have to upgrade my kernel because of that, 4.4.0 (Ubuntu's 4.4.0-36-generic x86_64).

Lnce, 2016-09-01:
Ok. I will try that as I have only tried with default Ubuntu kernels (4.1.x), which don't work for some reason.

brynnneilson, 2016-11-07:
I'm making progress. My 3210A now shows up when I run readall.py but I get all of these other errors after: 
```
python test.py 
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:recv: 0x1 0x2b 0xe 0x1 0x1 0x0 0x0 0x3 0x0 0x15 0x45 0x50 0x73 0x6f 0x6c 0x61 0x72 0x20 0x54 0x65 0x63 0x68 0x20 0x63 0x6f 0x2e 0x2c 0x20 0x4c 0x74 0x64 0x1 0xb 0x54 0x72 0x61 0x63 0x65 0x72 0x33 0x32 0x31 0x30 0x41 0x2 0xd 0x56 0x30 0x31 0x2e 0x31 0x32 0x2b 0x56 0x30 0x32 0x2e 0x31 0x31 0x7e 0x56
DEBUG:pymodbus.factory:Factory Response[43]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 1
{0: 'EPsolar Tech co., Ltd', 1: 'Tracer3210A', 2: 'V01.12+V02.11'}

<pyepsolartracer.registers.Register instance at 0xb62a00d0>
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x2 0x27 0x10 0xa3 0xc
DEBUG:pymodbus.factory:Factory Response[4]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 2
read_input_registers: 10000
DEBUG:pymodbus.transaction:Running transaction 3
DEBUG:pymodbus.transaction:recv: 0x1 0x83 0x2 0xc0 0xf1
DEBUG:pymodbus.factory:Factory Response[131]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 3

<pyepsolartracer.registers.Register instance at 0xb62a00f8>
DEBUG:pymodbus.transaction:Running transaction 4
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x2 0xb 0xb8 0xbe 0x72
DEBUG:pymodbus.factory:Factory Response[4]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 4
read_input_registers: 3000
DEBUG:pymodbus.transaction:Running transaction 5
DEBUG:pymodbus.transaction:recv: 0x1 0x83 0x2 0xc0 0xf1
DEBUG:pymodbus.factory:Factory Response[131]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 5

```

FritzBox360, 2017-02-21:
Hello, 

same problem here, i think,  with a Tracer 1210A. If i run readall.py i got this

`DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:recv: 0x1 0x2b 0xe 0x1 0x1 0x0 0x0 0x3 0x0 0x15 0x45 0x50 0x73 0x6f 0x6c 0x61 0x72 0x20 0x54 0x65 0x63 0x68 0x20 0x63 0x6f 0x2e 0x2c 0x20 0x4c 0x74 0x64 0x1 0xb 0x54 0x72 0x61 0x63 0x65 0x72 0x31 0x32 0x31 0x30 0x41 0x2 0xd 0x56 0x30 0x31 0x2e 0x31 0x32 0x2b 0x56 0x30 0x32 0x2e 0x31 0x31 0x7 0xb4
DEBUG:pymodbus.factory:Factory Response[43]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 1
{0: 'EPsolar Tech co., Ltd', 1: 'Tracer1210A', 2: 'V01.12+V02.11'}

<pyepsolartracer.registers.Register instance at 0x7f44b0ea8830>
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x2 0x27 0x10 0xa3 0xc
DEBUG:pymodbus.factory:Factory Response[4]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 2
read_input_registers: 10000
DEBUG:pymodbus.transaction:Running transaction 3
DEBUG:pymodbus.transaction:recv: 0x1 0x83 0x2 0xc0 0xf1
DEBUG:pymodbus.factory:Factory Response[131]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 3
read_holding_registers: Exception Response(131, 3, IllegalAddress)

<pyepsolartracer.registers.Register instance at 0x7f44b0ea8878>
DEBUG:pymodbus.transaction:Running transaction 4
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x2 0x3 0xe8 0xb9 0x8e
DEBUG:pymodbus.factory:Factory Response[4]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 4
read_input_registers: 1000
DEBUG:pymodbus.transaction:Running transaction 5
DEBUG:pymodbus.transaction:recv: 0x1 0x83 0x2 0xc0 0xf1
DEBUG:pymodbus.factory:Factory Response[131]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 5
read_holding_registers: Exception Response(131, 3, IllegalAddress)

[...]
`
Any ideas? Is it a problem in the script or with my configuration? 

Thank you

brynnneilson, 2017-02-21:
That looks correct to me. What issue are you having?

FritzBox360, 2017-02-21:
Yes, the first "output" with the model is correct. But all the other stuff is not here. 

Not sure but these `read_holding_registers: Exception Response(131, 3, IllegalAddress)` looks strange to me

brynnneilson, 2017-02-21:
Oh I see. Looks like you're getting most of the data back though. I ended up using this: https://github.com/kasbert/epsolar-tracer

FritzBox360, 2017-02-21:
Yes, this is the same tool i am using. That is why we are under this topic ;)

brynnneilson, 2017-02-21:
Oops. Sorry I linked to the wrong project. I meant this PHP one here: https://github.com/toggio/PhpEpsolarTracer  If you want my PHP file PM me and I'll send them to you.

FritzBox360, 2017-02-21:
How can i PM you although GitHub removed the private messaging feature? I am very new here - this was my first question

brynnneilson, 2017-02-21:
Here's the file: http://solar.spinningplanet.co.nz/PhpEpsolarTracer-backup.tar.gz

FritzBox360, 2017-02-23:
Thank you for the link. I have tried it but no luck with the Tracer 1210A - Do you have a Tracer A Series or another device?

brynnneilson, 2017-02-23:
I have a Tracer3210A

matthubbert100, 2017-02-23:
if you only want a few pieces of data from the controller (voltage, amps, etc) you can achieve it with very few lines of code and the pymodbus library. Have a look at my solar monitor - www.solarpoweredhome.co.uk

 It's down at the minute as my 2210A developed a short and stopped working :(

FritzBox360, 2017-02-24:
Thank you all!  I got the PHP-Scripts to work. It was my fault. There was a permission issue. 

You have to do `usermod -a -G dialout www-data` for adding www-data to access the serial device **AND** restart your system because without a restart the permission are not correct.

Lnce, 2017-02-26:
Matt: I tried your code but it returned this
  File "datalogger.py", line 67, in <module>
    sV = float(result.registers[0] / 100.0) # Solar voltage is register 3100, divide by 100
AttributeError: 'NoneType' object has no attribute 'registers'

Any ideas?

wrybread, 2017-03-11:
@matthubbert100: thank you! Just what I needed. Happily reading the status of my 4210A without changing a thing in your code.

NilanjPatel, 2017-06-30:
@brynnneilson @FritzBox360  i am getting error while reading holding register from schneider EM64300, using USB to RS485 adapter , that error is (131, 3, IllegalAddress) but i wire right address, i tried every different  way to put address but still not working.

kasbert, 2023-04-29:
The test program uses wrong methods. One should use either read_coils or read_discrete_inputs and the test program tries both.

closed 2023-04-29 by kasbert