kasbert/epsolar-tracer #12: problem with the driver

dutcou, 2017-01-11:
I wrote a datalogger in Python (using Pymodbus) which stores the data from my EPSolar Charge Controller in a SQLite3 database. The connection between my PC and the EP LS1024B is a EPSolar USB UART cable CC-USB-TTL-150. On my Windows-PC everything is working fine.
I try to port this to my RaspberryPi3 (linux version 4.4.38-v7+) but until now without success. I am not able to get a connection between my RPi and the LS1024. Based on a Internet search I presume that the problem is caused by a wrong driver. My RPi sees the cable on ttyACM0 and uses the cdc-acm driver. The search learned me that the xr_usb_serial_common is the correct driver so I downloaded the source files .
But I can not get a .ko file with the make command. I get as error "#include <linux/init.h> file or map doesn't exist". I installed raspberrypi-kernel-headers and copied all the files to several places but without success.
Can you explain me what to do to get a .ko file from the driver?

brynnneilson, 2017-01-16:
Here's how I got it working on my Asustor ARM based NAS: https://forum.asustor.com/viewtopic.php?f=28&t=8355&sid=bccd5e4da2dcb1023f9aa5d8ad81703d&start=10

You might also want to see if the driver I compiled will work with your RPi - probably won't. Download here: http://solar.spinningplanet.co.nz/xr_usb_serial_common.ko

brynnneilson, 2017-01-16:
This page also describes how to get it working in RPi: http://indilib.org/forum/general/1085-baader-diamond-steeldrive-not-working.html?start=12

kasbert, 2023-03-20:
Closing as stale

closed 2023-03-20 by kasbert