kasbert/epsolar-tracer #16: Not an issue just a query regarding Home Assistant

frank451a, 2017-06-06:
Hi

Is there anyway of integrating this into home assistant http://home-assistant.io/

Thanks

kasbert, 2017-06-06:
Currently no, but you are free to write a component.

GusPS, 2019-05-01:
Looking for the same thing as @frank451a, I found out that Home Assistant has the Modbus component: https://www.home-assistant.io/components/modbus/
With this, you can integrate the EPSolar registers into HA.

corvidus, 2020-01-26:
@GusPS, @frank451a et al, 
FWIW, I'm using a [PHP alternative](https://github.com/toggio/PhpEpsolarTracer) of this so the code isn't compatible, but I ship the output to Influx for long-term logging, then read it out into Grafana (for visualisation and analysis) and Home Assistant (for automation triggers). You could alternatively publish the output to MQTT and create an [MQTT Sensor](https://www.home-assistant.io/integrations/sensor.mqtt/) if you just want to manipulate it another way. 
Hope that helps.

closed 2023-03-20 by kasbert