kasbert/epsolar-tracer #20: Kernel 4.9 (raspberry pi) hangs on modbus connect

vrosca, 2017-09-08:
Haven't had time to dive deeper into the root cause - when trying to make a transaction on a Pi Zero W with kernel 4.9, the kernel immediately crashes. It doesn't even write a kernel panic to stdout or logs - hangs for a few seconds and then restarts. Has anyone else experienced this? With kernel 4.4 I had 100% CPU usage during the modbus transactions for the full 1+ second of delay between successive calls (tried setting timeout to a lower value, didn't help).

vrosca, 2017-09-09:
Update: problem doesn't occur on Arch linux kernel 4.9.43 - while it does happen on raspbian with kernel 4.9.43

vrosca, 2017-09-10:
Update 2: device is a Raspberry Pi Zero, problem occurs on both Arch an Raspbian (same kernel) when the dwc2 dtoverlay is enabled (for g_ether - gadget-mode USB). With dwc2/g_ether disabled, there are no issues

vrosca, 2017-09-14:
Update 3: in case anyone else runs into this problem - if you comment out the modprobe -r / modprobe commands for usbserial the issue does not seem to occur at all, even when configured with the dwc2 overlay + g_ether. It seems however to then take a few seconds after the insmod before the first values can be read (attempting to read 5-10 seconds after the insmod results in a 'None' response for all registers)

dmdelorme, 2017-12-23:
I found the interface flaky when I had a loose wire on any part of the system and time out needs to be about .2 
 `serialclient = ModbusClient(method = 'rtu', port = '/dev/ttyXRUSB0', baudrate = 115200, timeout=0.2)`

hoangv, 2021-03-13:
Got this to work in 2021 after some headache.  This will probably help someone else.  I resolved this (pymodbus read_device_info, timeout errors) by downgrading pymodbus to 1.2.0.

kasbert, 2023-03-20:
Current version is changed to use pymodbus 3.2

closed 2023-03-20 by kasbert