kasbert/epsolar-tracer #21: Oddities with the device

delovoy, 2017-09-19:
sudo insmod ./xr_usb_serial_common.ko

cat /etc/modprobe.d/bl.cdc_acm.conf 
blacklist cdc_acm 

tail -f / var / log / syslog -n 0 - [1.txt](https://github.com/kasbert/epsolar-tracer/files/1313936/1.txt)

help

delovoy, 2017-09-19:
cat /etc/issue.net 
Ubuntu 16.04.3 LTS

uname -a 
Linux IdeaPad 4.10.0-35-generic #39~16.04.1-Ubuntu SMP Wed Sep 13 09:02:42 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

I'm confused by errors, I did not check the module's functionality. Just do not want to screw up when programming the instruments. Are there any such mistakes?

kasbert, 2017-09-20:
It may still work. I wouldn't worry about the errors.
Modem manager tries to use it as a modem and does not have a configuration for it.
mtp-probe checks if it is a camera or phone or something.

closed 2017-09-23 by delovoy