kasbert/epsolar-tracer #25: Has anyone been able to get this driver installed on the pi?

pmitham, 2018-02-07:
I can't get this driver to install at all.

When I run make.  I get the following error.

pi@openplotter:~/xr_usb_serial_common-1a $ make
make -C /lib/modules/4.9.35-v7+/build M=/home/pi/xr_usb_serial_common-1a
make[1]: Entering directory '/lib/modules/4.9.35-v7+/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/lib/modules/4.9.35-v7+/build'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2

Do these drivers actually work??

pelom, 2018-02-14:
I'm also having this problem.
My system does not have the build folder
make[1]: Entering directory '/lib/modules/4.9.35-v7+/build'

pmitham, 2018-02-14:
I've since figured out that You need to download and install the raspbian
kernals in order to be able to compile the drivers.  The linux people out
there seem to forget that we all don;'t intrinsically know this! (specially
when you're new to Linux like I am)

before you do anything ( assuming you are on Raspbian)  Run this:

sudo apt-get install raspberrypi-kernel-headers
then you should be able to run the
sudo make
sudo insmod.  I also had to modify the modules file to include the driver
name  and black list the Aca-adm driver

Pierre

bonkas, 2018-10-18:
I seem to be stuck, have had this working for over a year with no issues but after having to reinstall my sd card I am not having any luck at all.

Have already done the above:

raspberrypi-kernel-headers is already the newest version (1.20180924-1).

But I get this when trying to build...

make[1]: Entering directory '/usr/src/linux-headers-4.14.71+'
make[1]: *** No rule to make target 'Files/Linux'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.14.71+'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2

bonkas, 2018-10-19:
Seems was an issue with my image. Blanked the SD and wrote ANOTHEr image to it, first thing I did was sudo apt-get install raspberrypi-kernel-headers

SUCCESS! Finally. Now to hopefully make some progress on this project :)

smros, 2018-12-09:
I managed to get this driver compiled after a long process.  I have the driver loaded, but by program can't seem to access it.  I had to manually load the driver with "sudo" at each step.  When I look at the results from lsmod, I see that both drivers are "used by" "0".  Does this mean only root can access it?  If so, is there a way to make the module and port available to all?

[image]

kasbert, 2023-03-20:
Closing as stale