kasbert/epsolar-tracer #26: Why do you have to reinstall these drivers after each reboot???

pmitham, 2018-02-08:
I've gone through the  install per the readme.txt file but when I re boot the pi  I have to re run the make and insmod commands each time in order for the pi to see theses USB devices.  There must be a step missing in the instructions!!  help!

jhulstedt, 2018-03-03:
What I had to do to get over this problem, these commands on my raspberry pi:
# This makes a link to the xr_usb_serial_common.ko file (notice the little tick marks on the `uname -r`)
sudo ln -s /path/to/your-kernel-module.ko /lib/modules/`uname -r`
# This checks dependencies May not have to do this but it did not hurt.
sudo depmod -a
# Installs the xr_usb_serial_common module replace "your-kernel-module" with "xr_usb_serial_common"
sudo modprobe your-kernel-module

I don't thing you should have to run make again because those files are there. But insmod commands are not persistent across boots or new sessions either I think.

Got this from https://www.computerhope.com/unix/modprobe.htm

jimlynnjulian, 2018-09-24:
I wrote and installed a service file to do everything and made sure the service runs before any other Epever related services.

jimlynnjulian, 2018-09-24:
[Unit]
Description=Epever Exar driver install Service
After=multi-user.target hwclock.service

[Service]
Type=oneshot
ExecStart=/bin/bash /home/pi/epdriver.sh

[Install]
WantedBy=multi-user.target

jimlynnjulian, 2018-09-24:
Run 
sudo systemctl enable epdriver.service
sudo systemctl start epdriver.service

jimlynnjulian, 2018-09-24:
#!/bin/bash

#version 1  -------------------
# Remove ttyACM0
sudo rmmod cdc-acm

sudo modprobe -r cdc-acm

# Insert ttyXRUSB0
sudo insmod /home/pi/xr_usb_serial_common.ko

# command to list usb driver modules:
#    ls /dev/tty*

kasbert, 2023-03-20:
Closing as stale

closed 2023-04-29 by kasbert