kasbert/epsolar-tracer #27: Driver for the XR21B1411

mcsarge, 2018-03-26:
I am running Raspberry Pi with Kernel 4.14.29-v7+ and I was able to get the 1b version of this driver from the Exar website since I am running a newer version of the kernel than the 1a version is for.

I am able to compile and install the driver and even able to load it for the device, having it create the /dev/ttyXRUSB0 device. But when I run the test programs, I get the following in dmesg - any advice?

[165443.591667] tty ttyXRUSB0: xr_usb_serial_tty_install
[165443.591733] tty ttyXRUSB0: xr_usb_serial_tty_open
[165443.591746] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_port_activate
[165443.592008] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_ctrl_msg - rq 0x22, val 0x3, len 0x0, result -32
[165443.592028] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_ctrl_irq - urb shutting down with status: -2
[165443.592589] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_set_reg Error:-32
[165443.592822] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_set_reg Error:-32
[165443.592945] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_set_reg Error:-32
[165443.593069] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_set_reg Error:-32
[165443.593194] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_set_reg Error:-32
[165443.593320] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_set_reg Error:-32
[165443.593445] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_set_reg Error:-32
[165449.607737] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_tty_close
[165449.607770] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_port_shutdown
[165449.608029] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_ctrl_msg - rq 0x22, val 0x0, len 0x0, result -32
[165449.608151] cdc_xr_usb_serial 1-1.4:1.0: xr_usb_serial_tty_cleanup

xoul, 2018-05-27:
Same problem here.

lzahradil, 2018-06-10:
Same problem

mcsarge, 2018-06-10:
Here is what I did. I bought a new RS485 USB interface from Amazon that uses the FTDI serial hardware and it works perfectly. http://a.co/9mGUuzH

jimlynnjulian, 2018-09-24:
I've recently developed a similar problem. A physical USB port stopped working. I moved the cable and that worked. Problem was I hjad to keep moving the cable. Added a thumb drive and no longer had to move the cable. Today, nothing works with th Exar driver. I workaround is nice, but I'd like to know what's going on.
I read there are polyfuses that trip and take a day or two to reset. That matches my situation.
Could the Epever Tracer be increasing the current due to a leaky component?
I'll leave my unit disonnected from the RPi for a few days and see if my USB ports are functional for the Tracer again. If so, I'm going to contact Epever and ask for a schematic or a maybe a replacement.
If the current draw is increasing gradually, the new cable suggested may be a temporary fix.

SCarns, 2022-07-05:
> Here is what I did. I bought a new RS485 USB interface from Amazon that uses the FTDI serial hardware and it works perfectly. http://a.co/9mGUuzH

Just wired ground, data+, data- and plug it in? FTDI is usually well supported. I'm having a hard time getting the xr_usb_serial drive to build and load. PITA. If the FTDI adapter works, then I'm all about it!

closed 2023-03-20 by kasbert