kasbert/epsolar-tracer #28: Write Code

theveng, 2018-04-21:
Hi, anyone succeded to write values ? I'd like to be able to control the load status. Thank you

kasbert, 2018-04-22:
Should work if the register is read-write.

theveng, 2018-04-22:
I know it's possible, was just looking for the python code to do it... thank you

mjkorhon, 2022-03-29:
Hi, I'm also having trouble writing Voltage settings. I can successfully change "Battery Type" to 0 (custom), but I have no luck in changing nearly anything else. For example trying to set "Low voltage disconnect" I get "Exception Response(144, 16, SlaveFailure)" from modbus. Any ideas?

gjmkoper, 2023-05-18:
Same here, exception probably means write protection. Is there maybe a special sequence to follow?