kasbert/epsolar-tracer #30: Problem: AttributeError: 'ModbusIOException' object has no attribute 'information'

borgelite, 2018-06-03:
I have a rpi3 and installed the driver.
dmesg:
usb 1-1.5: New USB device found, idVendor=04e2, idProduct=1411
[ 4386.853868] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4386.853877] usb 1-1.5: Product: XR21B1411
[ 4386.853885] usb 1-1.5: Manufacturer: Exar Corp.
[ 4386.853893] usb 1-1.5: SerialNumber: R3991242051
[ 4386.854816] cdc_xr_usb_serial 1-1.5:1.0: USB_device_id idVendor:04e2, idProduct 1411
[ 4386.854830] cdc_xr_usb_serial 1-1.5:1.0: This device cannot do calls on its own. It is not a modem.
[ 4386.854840] cdc_xr_usb_serial 1-1.5:1.0: The data interface has switched endpoints
[ 4386.854847] cdc_xr_usb_serial 1-1.5:1.0: interfaces are valid
[ 4386.854861] cdc_xr_usb_serial 1-1.5:1.0: epwrite->bEndpointAddress =1
[ 4386.854938] cdc_xr_usb_serial 1-1.5:1.0: ttyXR_USB_SERIAL0: USB XR_USB_SERIAL device
[ 4386.856227] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_ctrl_msg - rq 0x22, val 0x0, len 0x0, result 0
[ 4386.856620] cdc_xr_usb_serial 1-1.5:1.0: xr_usb_serial_ctrl_msg - rq 0x20, val 0x0, len 0x7, result 7

But when i execute i can not read.
/home/pi/epsolar-tracer# python readall.py
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x2b 0xe 0x1 0x0 0x70 0x77
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received))
DEBUG:pymodbus.framer.rtu_framer:Frame - [] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Traceback (most recent call last):
  File "readall.py", line 24, in <module>
    print repr(response.information)
AttributeError: 'ModbusIOException' object has no attribute 'information'

What is the Problem?

akissu, 2020-03-20:
I'm having the same issue. I'm thinking it's this:

```
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received))
```

Does pymodbus put serial devices into RS485 mode? I had this issue in Windows and explicitly setting RS485 mode fixed it.

kasbert, 2020-03-29:
If you use the modified Exar kernel module, it always puts the device into RS485 mode.

kasbert, 2023-03-20:
Closed as stale