kasbert/epsolar-tracer #32: Epever ET6415BND

seesix, 2018-09-10:
Hi there,
I am trying to run your code on a Pi Zero W with the latest version of Raspian Stretch

However, when I run this code

#################################

#!/usr/bin/env python
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
client = ModbusClient(method = 'rtu', port = '/dev/ttyAMA0', baudrate = 115200)
client.connect()
while True:
        result = client.read_input_registers(0x03, 1, unit=1)
# Do something with the data
print result
client.close()

###############################

it returns this error

###############################

Traceback (most recent call last):
  File "./test.py", line 9, in <module>
    result = client.read_input_registers(0x03, 1, unit=1)
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/client/common.py", line 125, in read_input_registers
    return self.execute(request)
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/client/sync.py", line 107, in execute
    return self.transaction.execute(request)
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/transaction.py", line 136, in execute
    full=full
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/transaction.py", line 202, in _transact
    size = self._send(packet)
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/transaction.py", line 219, in _send
    return self.client.framer.sendPacket(packet)
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/framer/rtu_framer.py", line 272, in sendPacket
    size = self.client.send(message)
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/client/sync.py", line 76, in send
    return self._send(request)
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/client/sync.py", line 491, in _send
    waitingbytes = self._in_waiting()
  File "/home/pi/.local/lib/python2.7/site-packages/pymodbus/client/sync.py", line 473, in _in_waiting
    waitingbytes = getattr(self.socket, in_waiting)()
  File "/home/pi/.local/lib/python2.7/site-packages/serial/serialutil.py", line 590, in inWaiting
    return self.in_waiting
  File "/home/pi/.local/lib/python2.7/site-packages/serial/serialposix.py", line 467, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
IOError: [Errno 25] Inappropriate ioctl for device

#############################

i'd be most grateful for any advise on how to solve this

kasbert, 2018-09-11:
It might be a problem with the Raspbian distibution.
Here is a similar case:  https://www.raspberrypi.org/forums/viewtopic.php?t=218469

seesix, 2018-09-13:
I just tried with a fresh install of  Jessie and same error. My guess is, its the cable, I'm using the one supplied by Epever, have order a new one.

seesix, 2018-09-26:
I have just tried again with  a new FTDI cable and it returns this error.

####################################################

 sudo ./info.py 
Traceback (most recent call last):
  File "./info.py", line 18, in <module>
    client.connect()
  File "/home/pi/battery/epsolar-tracer/pyepsolartracer/client.py", line 33, in connect
    return self.client.connect()
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/client/sync.py", line 451, in connect
    parity=self.parity)
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 216, in __init__
    self.port = port
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 260, in port
    raise ValueError('"port" must be None or a string, not {}'.format(type(port)))
ValueError: "port" must be None or a string, not <type 'int'>

################################################

Any ideas on how to fix this would be gratefully received.

Thanks

kasbert, 2018-09-26:
Edit info.py and set correct arguments:

serialclient = ModbusClient(method='rtu', port='/dev/ttyXRUSB0', baudrate=115200, stopbits = 1, bytesize = 8, timeout=1)

Do you see the wanted the /dev/tty* device ?

seesix, 2018-09-28:
thank you so much for the help.

in the end my device was ttyUSB1

seesix, 2018-10-25:
hey kasbert

i'm still stuck on this, when i run info.py i get this
=====================================================
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated input voltage'
Charging equipment rated input voltage = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated input current'
Charging equipment rated input current = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated input power'
Charging equipment rated input power = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated input power L'
Charging equipment rated input power L = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated input power H'
Charging equipment rated input power H = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated output voltage'
Charging equipment rated output voltage = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated output current'
Charging equipment rated output current = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated output power'
Charging equipment rated output power = None
Charging equipment rated output power L = 378.56W
INFO:pyepsolartracer.registers:No value for register 'Charging equipment rated output power H'
Charging equipment rated output power H = None
INFO:pyepsolartracer.registers:No value for register 'Charging mode'
Charging mode = None
INFO:pyepsolartracer.registers:No value for register 'Rated output current of load'
Rated output current of load = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment input voltage'
Charging equipment input voltage = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment input current'
Charging equipment input current = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment input power'
Charging equipment input power = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment input power L'
Charging equipment input power L = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment input power H'
Charging equipment input power H = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment output voltage'
Charging equipment output voltage = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment output current'
Charging equipment output current = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment output power'
Charging equipment output power = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment output power L'
Charging equipment output power L = None
INFO:pyepsolartracer.registers:No value for register 'Charging equipment output power H'
Charging equipment output power H = None
INFO:pyepsolartracer.registers:No value for register 'Discharging equipment output voltage'
Discharging equipment output voltage = None
INFO:pyepsolartracer.registers:No value for register 'Discharging equipment output current'
Discharging equipment output current = None
INFO:pyepsolartracer.registers:No value for register 'Discharging equipment output power'
Discharging equipment output power = None
INFO:pyepsolartracer.registers:No value for register 'Discharging equipment output power L'
Discharging equipment output power L = None
INFO:pyepsolartracer.registers:No value for register 'Discharging equipment output power H'
Discharging equipment output power H = None
INFO:pyepsolartracer.registers:No value for register 'Battery Temperature'
Battery Temperature = None
INFO:pyepsolartracer.registers:No value for register 'Temperature inside equipment'
Temperature inside equipment = None
INFO:pyepsolartracer.registers:No value for register 'Power components temperature'
Power components temperature = None
INFO:pyepsolartracer.registers:No value for register 'Battery SOC'
Battery SOC = None
INFO:pyepsolartracer.registers:No value for register 'Remote battery temperature'
Remote battery temperature = None
INFO:pyepsolartracer.registers:No value for register "Battery's real rated power"
Battery's real rated power = None
[... 226 lines cut ...]
#response = client.read_holding_registers(0, 1, unit=1)
#response = client.read_input("Charging equipment rated input voltage")
print str(response)

for reg in registers:
    #print
    #print reg
    value = client.read_input(reg.name)
    print value
    #if value.value is not None:
    #    print client.write_output(reg.name,value.value)

for reg in coils:
    #print
    #print reg
    value = client.read_input(reg.name)
    print value
    #print client.write_output(reg.name,value.value)

client.close()

===============================================

Any ideas what I'm doing wrong?
Thanks again for the help

kasbert, 2023-03-20:
Closed as stale

closed 2023-03-20 by kasbert