kasbert/epsolar-tracer #36: AttributeError: 'ModbusIOException' object has no attribute 'registers'

yogawenno, 2019-07-29:
hey guys, please help me to solve this problem
i have code

from pymodbus.client.sync import ModbusSerialClient as ModbusClient

client = ModbusClient(method = 'rtu', port = '/dev/ttyACM0', baudrate = 115200)
client.connect()

result = client.read_input_registers(0x3100,6,unit=1)
solarVoltage = float(result.registers[0] /100.0)
solarCurrent = float(result.registers[1] /100.0)
batteryVoltage = float(result.registers[4] /100.0)
chargeCurrent = float(result.registers[5] /100.0)

print("solar voltage : ",solarVoltage)
print("solar current : ",solarCurrent)
print("battery volatage : ",batteryVoltage)
print("charge current : ",chargeCurrent)
client.close()

when i run this code, i get error like this

Traceback (most recent call last):
  File "yutub.py", line 7, in <module>
    solarVoltage = float(result.registers[0] /100.0)
AttributeError: 'ModbusIOException' object has no attribute 'registers'

can anybody can help me to fix this problem?
thanks :)

bistory, 2020-03-28:
Same problem here, by displaying modbus error, I get this message :

Incomplete message received, expected at least 2 bytes

I updated my Debian but changed nothing else electrically speaking...

kasbert, 2020-03-29:
I think the driver for ttyACM0 cannot put the device to RS485 mode.

bistory, 2020-03-29:
I finally fixed the problem by using the xr driver version 1a instead of 1c... strange ! (I use Dumasto install it)

quantumbitflip, 2020-06-02:
@bistory Looking into your solution using the specific driver version, but what is "Dumasto" I cannot find a single reference to it.
Thank you

MichaelErmer, 2020-07-09:
> I finally fixed the problem by using the xr driver version 1a instead of 1c... strange ! (I use Dumasto install it)

Same here 1A which is in the repo works fine, 1C from the vendor website does not work at all.

Ansari28, 2022-07-27:
> I finally fixed the problem by using the xr driver version 1a instead of 1c... strange ! (I use Dumasto install it)

Can you please Elaborate more about this. i don't understand "  by using the xr driver version 1a instead of 1c " 
Traceback (most recent call last):
  File "/home/pi/Desktop/test.py", line 17, in <module>
    print(result.registers)
AttributeError: 'ModbusIOException' object has no attribute 'registers'
pi@raspberrypi:~/Desktop $ 
 
Facing this error from a long time but unable to understand what exactly mean it or how can i resolve this error???/

bistory, 2022-07-27:
> > I finally fixed the problem by using the xr driver version 1a instead of 1c... strange ! (I use Dumasto install it)
> 
> Can you please Elaborate more about this. i don't understand " by using the xr driver version 1a instead of 1c " Traceback (most recent call last): File "/home/pi/Desktop/test.py", line 17, in print(result.registers) AttributeError: 'ModbusIOException' object has no attribute 'registers' pi@raspberrypi:~/Desktop $
> 
> Facing this error from a long time but unable to understand what exactly mean it or how can i resolve this error???/

I used driver from the manufacturer that didn't work. The one on this repo works great. But your problem is maybe not related to mine ;-)

Ansari28, 2022-07-27:
Ok. Thanks for your response. if you know any solution for this error please let me know.

kasbert, 2023-03-20:
I guess this was resolved.