kasbert/epsolar-tracer #40: ERROR: could not insert module ./xr_usb_serial_common.ko: Invalid module format

kevinkey619, 2021-09-11:
I keep getting the below error:

sudo insmod ./xr_usb_serial_common.ko
insmod: ERROR: could not insert module ./xr_usb_serial_common.ko: Invalid module format

kasbert, 2021-09-11:
does "dmesg" command reveal anything ?

kevinkey619, 2021-09-11:
Thanks for the quick reply. I'm fairly technically savvy, but I'm more of a DOS/Windows person. Not a Linux expert. Here you go:

pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1450 SMP Wed Sep 8 14:29:23 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2133 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 878804K/970752K available (10240K kernel code, 1312K rwdata, 2948K rodata, 1024K init, 863K bss, 26412K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32055 entries in 63 pages
[    0.000000] ftrace: allocated 63 pages with 6 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3ac/0x580 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000024] Switching to timer-based delay loop, resolution 52ns
[    0.000308] Console: colour dummy device 80x30
[    0.001176] printk: console [tty1] enabled
[    0.001247] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001305] pid_max: default: 32768 minimum: 301
[    0.001523] LSM: Security Framework initializing
[    0.001776] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001826] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003275] cgroup: Disabling memory control group subsystem
[    0.003553] CPU: Testing write buffer coherency: ok
[    0.004049] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005292] Setting up static identity map for 0x100000 - 0x10003c
[    0.005499] rcu: Hierarchical SRCU implementation.
[    0.006424] smp: Bringing up secondary CPUs ...
[    0.007576] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008828] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.010121] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.010279] smp: Brought up 1 node, 4 CPUs
[    0.010332] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.010363] CPU: All CPU(s) started in HYP mode.
[    0.010390] CPU: Virtualization extensions available.
[    0.011371] devtmpfs: initialized
[    0.029015] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[... 271 lines cut ...]
[   14.184920] uart-pl011 3f201000.serial: no DMA platform data
[   14.329068] 8021q: 802.1Q VLAN Support v1.8
[   15.074548] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   15.149925] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[... cut ...]
[   19.289127] Bluetooth: HCI UART protocol H4 registered
[   19.289269] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   19.289600] Bluetooth: HCI UART protocol Broadcom registered
[   19.806656] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   19.806680] Bluetooth: BNEP filters: protocol multicast
[   19.806714] Bluetooth: BNEP socket layer initialized
[   19.899482] Bluetooth: RFCOMM TTY layer initialized
[   19.899537] Bluetooth: RFCOMM socket layer initialized
[   19.899574] Bluetooth: RFCOMM ver 1.11
pi@raspberrypi:~ $

kasbert, 2021-09-12:
Try:
        To remove the CDC-ACM driver and install the driver:

        # rmmod cdc-acm
        # modprobe -r usbserial
        # modprobe usbserial
        # insmod ./xr_usb_serial_common.ko

# file ./xr_usb_serial_common.ko 
./xr_usb_serial_common.ko: ELF 32-bit LSB relocatable, ARM, EABI5 version 1 (SYSV), BuildID[sha1]=9da89b6ed09912808a7ae9de591584fd1c1bbb5f, not stripped

kevinkey619, 2021-09-12:
Thanks, but no luck. I still end up with the same error message even after performing all 4 steps:

 sudo insmod ./xr_usb_serial_common.ko
insmod: ERROR: could not insert module ./xr_usb_serial_common.ko: Invalid module format

kevinkey619, 2021-09-13:
My Raspberry Pi micro SD card died last night. This might have been the problem. Going to get a new card and reinstall everything. I'll let you know if the problem goes away.

kevinkey619, 2021-09-14:
Yeah, the problem was with the failing micro SD card on my Raspberry Pi. New card, reinstall, and all is well now.

closed 2021-09-14 by kevinkey619