kasbert/epsolar-tracer #41: fail to make driver

MohammadPCh, 2022-02-24:
It has these errors: 
```
make -C /lib/modules/5.13.0-30-generic/build M=/home/lobater/projects/test/epsolar-tracer/linux_usb_serial
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-30-generic'
  CC [M]  /home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.o
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c: In function ‘xr_set_flow_mode’:
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:636:27: error: expected ‘;’ before ‘int’
  636 |  unsigned long rs485_flags
      |                           ^
      |                           ;
  637 |  int ret;
      |  ~~~                       
In file included from ./include/linux/kernel.h:16,
                 from /home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:15:
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:653:33: error: ‘flags’ undeclared (first use in this function)
  653 |  spin_lock_irqsave(&data->lock, flags);
      |                                 ^~~~~
./include/linux/typecheck.h:11:9: note: in definition of macro ‘typecheck’
   11 |  typeof(x) __dummy2; \
      |         ^
./include/linux/spinlock.h:384:2: note: in expansion of macro ‘raw_spin_lock_irqsave’
  384 |  raw_spin_lock_irqsave(spinlock_check(lock), flags); \
      |  ^~~~~~~~~~~~~~~~~~~~~
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:653:2: note: in expansion of macro ‘spin_lock_irqsave’
  653 |  spin_lock_irqsave(&data->lock, flags);
      |  ^~~~~~~~~~~~~~~~~
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:653:33: note: each undeclared identifier is reported only once for each function it appears in
  653 |  spin_lock_irqsave(&data->lock, flags);
      |                                 ^~~~~
./include/linux/typecheck.h:11:9: note: in definition of macro ‘typecheck’
   11 |  typeof(x) __dummy2; \
      |         ^
./include/linux/spinlock.h:384:2: note: in expansion of macro ‘raw_spin_lock_irqsave’
  384 |  raw_spin_lock_irqsave(spinlock_check(lock), flags); \
      |  ^~~~~~~~~~~~~~~~~~~~~
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:653:2: note: in expansion of macro ‘spin_lock_irqsave’
  653 |  spin_lock_irqsave(&data->lock, flags);
      |  ^~~~~~~~~~~~~~~~~
./include/linux/typecheck.h:12:18: warning: comparison of distinct pointer types lacks a cast
   12 |  (void)(&__dummy == &__dummy2); \
      |                  ^~
./include/linux/spinlock.h:251:3: note: in expansion of macro ‘typecheck’
  251 |   typecheck(unsigned long, flags); \
      |   ^~~~~~~~~
./include/linux/spinlock.h:384:2: note: in expansion of macro ‘raw_spin_lock_irqsave’
  384 |  raw_spin_lock_irqsave(spinlock_check(lock), flags); \
      |  ^~~~~~~~~~~~~~~~~~~~~
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:653:2: note: in expansion of macro ‘spin_lock_irqsave’
  653 |  spin_lock_irqsave(&data->lock, flags);
      |  ^~~~~~~~~~~~~~~~~
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:654:2: error: ‘rs485_flags’ undeclared (first use in this function); did you mean ‘rseq_flags’?
  654 |  rs485_flags = data->rs485.flags;
      |  ^~~~~~~~~~~
      |  rseq_flags
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c: In function ‘xr_port_remove’:
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:1062:9: error: ‘return’ with a value, in function returning void [-Werror=return-type]
 1062 |  return 0;
      |         ^
/home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.c:1056:13: note: declared here
 1056 | static void xr_port_remove(struct usb_serial_port *port)
      |             ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/lobater/projects/test/epsolar-tracer/linux_usb_serial/xr_serial.o] Error 1
make[1]: *** [Makefile:1879: /home/lobater/projects/test/epsolar-tracer/linux_usb_serial] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-30-generic'
make: *** [Makefile:9: all] Error 2

```

how can I fix it?

kasbert, 2022-02-25:
I made a fix, please pull and try avain.

Strange, I could have sworn, that I compiled it.

kasbert, 2022-04-19:
I suppose this is ok now

closed 2022-04-19 by kasbert