kasbert/epsolar-tracer #43: failed to make driver

adadrag, 2022-04-18:
it is still failing to make the driver:

pi@raspberrypi4:~/xr_usb_serial_common-1a $ sudo make
make -C /lib/modules/5.15.32-v7l+/build M=/home/pi/xr_usb_serial_common-1a
make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v7l+'
  CC [M]  /home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.o
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_ctrl_irq’:
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:261:21: warning: unused variable ‘tty’ [-Wunused-variable]
  261 |  struct tty_struct *tty;
      |                     ^~~
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_process_read_urb’:
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:393:21: warning: unused variable ‘tty’ [-Wunused-variable]
  393 |  struct tty_struct *tty;
      |                     ^~~
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_softint’:
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:468:21: warning: unused variable ‘tty’ [-Wunused-variable]
  468 |  struct tty_struct *tty;
      |                     ^~~
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_reset_resume’:
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:1626:21: warning: unused variable ‘tty’ [-Wunused-variable]
 1626 |  struct tty_struct *tty;
      |                     ^~~
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c: At top level:
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:1691:17: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1691 |  .write_room =  xr_usb_serial_tty_write_room,
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:1691:17: note: (near initialization for ‘xr_usb_serial_ops.write_room’)
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:1695:21: error: initialization of ‘unsigned int (*)(struct tty_struct *)’ from incompatible pointer type ‘int (*)(struct tty_struct *)’ [-Werror=incompatible-pointer-types]
 1695 |  .chars_in_buffer = xr_usb_serial_tty_chars_in_buffer,
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:1695:21: note: (near initialization for ‘xr_usb_serial_ops.chars_in_buffer’)
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_init’:
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:1709:29: error: implicit declaration of function ‘alloc_tty_driver’ [-Werror=implicit-function-declaration]
 1709 |  xr_usb_serial_tty_driver = alloc_tty_driver(XR_USB_SERIAL_TTY_MINORS);
      |                             ^~~~~~~~~~~~~~~~
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:1709:27: warning: assignment to ‘struct tty_driver *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 1709 |  xr_usb_serial_tty_driver = alloc_tty_driver(XR_USB_SERIAL_TTY_MINORS);
      |                           ^
/home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.c:1726:3: error: implicit declaration of function ‘put_tty_driver’ [-Werror=implicit-function-declaration]
 1726 |   put_tty_driver(xr_usb_serial_tty_driver);
      |   ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /home/pi/xr_usb_serial_common-1a/xr_usb_serial_common.o] Error 1
make[1]: *** [Makefile:1868: /home/pi/xr_usb_serial_common-1a] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v7l+'
make: *** [Makefile:9: all] Error 2

kasbert, 2022-04-19:
Please pull and try again. I merged a pull request to fix compilation for 5.13 kernels and above

adadrag, 2022-04-19:
Perfect, it works now!

closed 2022-04-20 by kasbert