kasbert/epsolar-tracer #44: USB Fails To compile on Raspberry Pi  3 - 5.15.45-v8+

mitchross, 2022-06-09:
``` shell
vanillax@raspberrypi:~ $ git clone https://github.com/kasbert/epsolar-tracer.git
Cloning into 'epsolar-tracer'...
remote: Enumerating objects: 163, done.
remote: Counting objects: 100% (55/55), done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 163 (delta 29), reused 38 (delta 19), pack-reused 108
Receiving objects: 100% (163/163), 665.92 KiB | 1.84 MiB/s, done.
Resolving deltas: 100% (83/83), done.
vanillax@raspberrypi:~ $ ls
Bookshelf  Desktop  Documents  Downloads  epsolar-tracer  Music  Pictures  Public  Templates  Videos
vanillax@raspberrypi:~ $ cd epsolar-tracer/
vanillax@raspberrypi:~/epsolar-tracer $ ls
archive  img  info.py  LICENSE  linux_usb_serial  pyepsolartracer  readall.py  README.md  test  xr_usb_serial_common-1a
vanillax@raspberrypi:~/epsolar-tracer $ cd xr_usb_serial_common-1a/
vanillax@raspberrypi:~/epsolar-tracer/xr_usb_serial_common-1a $ ls
dkms.conf  Makefile  README.txt  xr_usb_serial_common.c  xr_usb_serial_common.h  xr_usb_serial_hal.c  xr_usb_serial_ioctl.h
vanillax@raspberrypi:~/epsolar-tracer/xr_usb_serial_common-1a $ make
make -C /lib/modules/5.15.45-v8+/build M=/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a
make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v8+'
  CC [M]  /home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.o
/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_ctrl_irq’:
/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.c:261:21: warning: unused variable ‘tty’ [-Wunused-variable]
  261 |  struct tty_struct *tty;
      |                     ^~~
/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_process_read_urb’:
/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.c:393:21: warning: unused variable ‘tty’ [-Wunused-variable]
  393 |  struct tty_struct *tty;
      |                     ^~~
/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_softint’:
/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.c:468:21: warning: unused variable ‘tty’ [-Wunused-variable]
  468 |  struct tty_struct *tty;
      |                     ^~~
/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.c: In function ‘xr_usb_serial_reset_resume’:
/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.c:1634:21: warning: unused variable ‘tty’ [-Wunused-variable]
 1634 |  struct tty_struct *tty;
      |                     ^~~
  MODPOST /home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/Module.symvers
  CC [M]  /home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.mod.o
  LD [M]  /home/vanillax/epsolar-tracer/xr_usb_serial_common-1a/xr_usb_serial_common.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v8+'
vanillax@raspberrypi:~/epsolar-tracer/xr_usb_serial_common-1a $ sudo ./xr_usb_serial_common.ko
sudo: ./xr_usb_serial_common.ko: command not found
vanillax@raspberrypi:~/epsolar-tracer/xr_usb_serial_common-1a $ ls
dkms.conf      Module.symvers          xr_usb_serial_common.h    xr_usb_serial_common.mod.c  xr_usb_serial_hal.c
Makefile       README.txt              xr_usb_serial_common.ko   xr_usb_serial_common.mod.o  xr_usb_serial_ioctl.h
modules.order  xr_usb_serial_common.c  xr_usb_serial_common.mod  xr_usb_serial_common.o
vanillax@raspberrypi:~/epsolar-tracer/xr_usb_serial_common-1a $ sudo insmod xr_usb_serial_common.ko
insmod: ERROR: could not insert module xr_usb_serial_common.ko: Invalid module format
vanillax@raspberrypi:~/epsolar-tracer/xr_usb_serial_common-1a $

```

kasbert, 2022-06-12:
Have you updated packages and not booted yet ?

```
make -C /lib/modules/5.15.45-v8+/build M=/home/vanillax/epsolar-tracer/xr_usb_serial_common-1a
                                           ^^ 
make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v8+'
                                                                                         ^^
```

zeeja, 2022-12-14:
Hi, I could not able to find the 10 amp [Tracer 1210RN] Maximum Power Point Tracking (MPPT) solar charge controller. Could you give me a suggestion to implement this project with a different possible available MPPT solar charge controller?

I found this one but it is Tracer4210AN:
https://www.amazon.de/EPEVER-Laderegler-controller-automatische-Hintergrundbeleuchtung/dp/B074YJ8NVT/ref=asc_df_B074YJ8NVT/?tag=googshopde-21&linkCode=df0&hvadid=256305048636&hvpos=&hvnetw=g&hvrand=13670960850257542562&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9044730&hvtargid=pla-432186488045&th=1

kasbert, 2022-12-14:
@zeeja [Tracer 1210RN] does not work with this software. Use https://github.com/xxv/tracer instead.
See also https://www.john.geek.nz/2014/09/ep-solar-mppt-tracer-charge-controller-serial-interface/

zeeja, 2022-12-19:
@kasbert Thank you for your reply and for sharing the information.

kasbert, 2023-03-20:
I guess this was resolved

closed 2023-03-20 by kasbert