kasbert/epsolar-tracer #50: cdc-acm

pterodyne, 2023-03-04:
I have the standard epever rs485 cable, and it worked fine in windows. But I switched over to linux and can't work this out.  The main issue is an extremely important piece of equipment also uses cdc-acm.  So Im under the impression that if I blacklist that driver that other device won't work.  Is there anything else I can do?  I saw another post where someone posted a different RS-485 to USB unit that uses FTDI.  But in that case Id need to make a cable so.  In addition to the first question above, anyone have the pinout of the RJ45 jack?

Thanks.

Bryan

pterodyne, 2023-03-04:
I found this on the web, from a github schematic for making a RS485 to wifi.  Id love to confirm if anyone knows:

Pins 1 and 2 V, Pins 3 and 4 data B, pins 4 and 6 Data A, and pins 7 and 8 Gnd.

Im guessing if I got this one I could just strip an ethernet cable

https://www.amazon.com/GearMo-Converter-Terminals-Windows-Support/dp/B005CPI0JQ/ref=sr_1_11?crid=16V034QVDLVLB&keywords=RS+485+to+USB+ftdi&qid=1677900694&refinements=p_85%3A2470955011&rnid=2470954011&rps=1&sprefix=rs485+to+usb+ftdi%2Caps%2C182&sr=8-11

And Im also guessing I wouldn't hook up power, Just A, B and Gnd.  I realize this isn't directly related to this project, but it seemed like someone here would know.

kasbert, 2023-03-07:
Try to unbind only the Exar device:
https://lwn.net/Articles/143397/

Googling will reveal two different pinouts and I am not sure which one is correct. I don't have a Tracer at hand, but I have the RS-485 dongle. I'll try to figure it out.

kasbert, 2023-03-12:
@pterodyne I added a wiring chapter to README. I think you should be able to use a cheaper half duplex RS-485 adapter. Use pins 4 and 6 for D- and D+.

pterodyne, 2023-03-13:
Awesome!  I have a FTDI to RS485 with screw terminals I can try just to make things easier.  It's good to know that one can unbind one device, but I think that the FTDI will be easier.  I also now have ordered an EPEVER BLE adapter, so I will probably go that route.  Thanks for all the support!

closed 2023-03-13 by pterodyne