kasbert/epsolar-tracer #53: Kernel header not found

kingfash, 2023-03-23:
Good day,

Please I need help in fixing the error below. I got it while I run: # dkms build -m xr_usb_serial_common -v 1a

"Error! echo
Your kernel headers for kernel 6.1.19-v8+ cannot be found at
/lib/modules/6.1.19-v8+/build or /lib/modules/6.1.19-v8+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located."

I have ran update. Please I don't know how i fix this.

Thank you

kasbert, 2023-03-23:
Does command "uname -r" return "6.1.19-v8+" ?
If not, reboot you machine.

Install kernel headers, if you alaready haven't.
sudo apt install linux-headers-$(uname -r)

kingfash, 2023-03-23:
Yes "uname-r" returned "6.1.19v8+

The driver was working b4 until I did an update using GUi over the weekend which upgraded the Kernel header.

bistory, 2023-09-29:
> Yes "uname-r" returned "6.1.19v8+
> 
> The driver was working b4 until I did an update using GUi over the weekend which upgraded the Kernel header.

I had the same problem, headers were not pushed by debian team, this should be resolved now

kasbert, 2025-01-15:
Upgrade to the latest Raspbian. It should work out of the box without xr_usb_serial_common driver.

closed 2025-01-15 by kasbert