kasbert/epsolar-tracer #54: EPEVER BLE-RJ45-A

pterodyne, 2023-03-30:
Let me start by saying I'm an idiot, and my parents are probably idiots too.  I bought the RS-485 adapter with FTDI, but then also bought the EPEVER BLE-RJ45-A.  I decided to try that instead figuring I could just map it to a rfcomm port like other devices I have and it should just work with this and other projects.  Well, not so fast. apparently it is BLE (even says it in the name, see my first comment).  BLE is, as I'm learning, a different animal.  Anyone play with this? I can map a rfcomm port, but it doesn't seem to work.  Looking at the BLE paring/reading/writing info there is a whole bunch of differences that may make it a non-starter without a specific project like this one to make it work, and I lack the skills/time to create one for myself.

Anyway, thanks for looking, and you can close this out without comment if you want. I just am hoping someone has a solution!  I also have the USB/RS485FTDI unit but I already installed the BLE and am 2 hours away from the Observatory hosting the EPEVER Solar charge controller. I do however have remote access VNC/ssh.

pterodyne, 2023-03-30:
Oh, and I did spend a couple hours dinking with https://github.com/Jakeler/ble-serial
seems like the way to go but I never got it to work.  Maybe I should just post there :)

pterodyne, 2023-03-30:
just a little more info.. If I use the ble-serial, here is the output:

`11:52:36.631 | INFO | linux_pty.py: Port endpoint created on /tmp/ttyBLE -> /dev/pts/1
11:52:36.631 | INFO | ble_interface.py: Receiver set up
11:52:36.870 | INFO | ble_interface.py: Trying to connect with 60:98:66:DC:54:91: BT-TH-66DC5491    
11:52:37.344 | INFO | ble_interface.py: Device 60:98:66:DC:54:91 connected
11:52:37.345 | INFO | ble_interface.py: Found write characteristic 0000ffd3-0000-1000-8000-00805f9b34fb (H. 35)
11:52:37.345 | INFO | ble_interface.py: Found notify characteristic 0000fff1-0000-1000-8000-00805f9b34fb (H. 45)
11:52:37.369 | INFO | main.py: Running main loop!
`
but if I try to run

epevermodbus --portname /dev/pts/2 --slaveaddress 1
Real Time Data
              1?61?61?61?61?6

This is the response, then it sort of locks up the terminal.  Maybe it's doing 232 instead of 485.  Dunno.  I'm a hack at best stringing github projects together with more tenacity than understanding.

Note that the actual port it reports is /dev/pts/1, but I get the normal set of errors when it cannot connect to anything on that.  It also creates /dev/pts/2 and 3 at the same time.

Bryan

kasbert, 2023-04-04:
I have only used Linux blueman to connect bluetooth serial ports. Tend to be quite unreliable, though.

pterodyne, 2023-04-18:
thanks. Ive wired up my ftdi 4s485 and need to just go install it.

closed 2023-04-18 by pterodyne