kasbert/epsolar-tracer #57: /dev/ttyXRUSBx doesn't exist anymore in 6.1.0-0.deb11.11-amd64

bistory, 2023-09-29:
Hi,
First thanks a lot for this driver, this works for years without any problem ;-)
I recently updated my kernel to 6.1.0-0.deb11.11-amd64 but the system doesn't create /dev/ttyXRUSB[1-3] files.

I ran dmesg to see what's going on, I get this : 

```
[    7.934377] xr_usb_serial_common: loading out-of-tree module taints kernel.
[    7.934504] xr_usb_serial_common: module verification failed: signature and/or required key missing - tainting kernel
[    7.987885] usbcore: registered new interface driver xr_serial
[    7.987924] usbserial: USB Serial support registered for xr_serial
[    7.989945] xr_serial 1-1.2:1.0: xr_serial converter detected
[    7.993852] usb 1-1.2: xr_serial converter now attached to ttyUSB0
[    7.993982] xr_serial 1-1.3:1.0: xr_serial converter detected
[    8.097843] usb 1-1.3: xr_serial converter now attached to ttyUSB1
[    8.098668] usbcore: registered new interface driver cdc_xr_usb_serial
[    8.098678] xr_usb_serial_common: Exar USB UART (serial port) driver
```

It seems it fallbacks to an other module (?) and I'm not able to use your scripts with ttyUSB0 and ttyUSB1

I tried removing the whole xr_usb_serial_common tree from DKMS and then reinstalling it, but it doesn't help.
I also tried the most recent xr_usb_serial_common v1d, backporting your fixes. The files are created but they return an empty string...

Unfortunately, I'm stuck, maybe there is a new driver to add in a blocklist ?

Thanks for your help ;-)

bistory, 2023-09-29:
Oh I found the solution : add xr_serial to blocklist :-)
Is there a way to use that mainline driver instead of compiling xr_usb_serial_common each time a new kernel pops ?

mitchross, 2023-11-12:
> Reference in new i

how did you add to block list?

bistory, 2023-11-12:
> > Reference in new i
> 
> how did you add to block list?

Like you did for cdc-acm (it's in the documentation) ;-)

kasbert, 2023-11-12:
The Linux 6.6 kernel has the needed code. If you are able to upgrade to that, you won't need this separate driver.

mitchross, 2023-11-12:
> The Linux 6.6 kernel has the needed code. If you are able to upgrade to that, you won't need this separate driver.

I'm using a Rpi3. Think it's 6.1 kernal still.

JanusHL, 2023-12-18:
Hi,
same issue with Armbian bullseye / Linux 6.1.63-current-meson64.
The system assign ttyUSB0 when plug the adapter.
I tried the 'bistory' blacklist solution ... but now don't show any ttyUSB0 nor ttyXRUSBx...
lsusb shows:
Bus 001 Device 007: ID 04e2:1411 Exar Corp. XR21B1411

But in dmesg did not appear to connect it:
[    3.652709] usb 1-1.2: New USB device found, idVendor=04e2, idProduct=1411, bcdDevice= 0.01
[    3.652731] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.652738] usb 1-1.2: Product: XR21B1411
[    3.652743] usb 1-1.2: Manufacturer: Exar Corp.
[    3.652748] usb 1-1.2: SerialNumber: R4326070561
[    3.791387] usb 1-1.3: new full-speed USB device number 5 **using xhci-hcd**

It seems to is using xhci-hcd like the USB wifi dongle....

bistory, 2023-12-18:
> The Linux 6.6 kernel has the needed code. If you are able to upgrade to that, you won't need this separate driver.

I fixed it by upgrading to a 6.6 kernel backport, thanks for the fix !

closed 2023-12-18 by bistory