kasbert/epsolar-tracer #58: Not working with native xr-serial from Linux

bistory, 2024-01-03:
Hi,

I'm having issues with this package for a while : I'm not able to get any data from my tracers...

I'm using Linux 6.6, the latest commit of this module in a container using pymodbus-3.6.2, pyserial-3.5 and pytz-2023.3.post1

And I get this error when I try the info script : 

```
python3 /epsolar/epsolartracer/info.py
WARNING:pymodbus.logging:Cleanup recv buffer before send: 0x0
Traceback (most recent call last):
  File "/epsolar/epsolartracer/info.py", line 28, in <module>
    response = client.read_device_info()
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/epsolar/epsolartracer/pyepsolartracer/client.py", line 48, in read_device_info
    raise response
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 4 bytes (2 received)
```

I used Exar driver before but I need cdc-acm module for a Zigbee receiver, so I'm stuck on that problem ^^
Do you have the same issue or do you have a solution?

Thanks !

JanusHL, 2024-01-03:
Well, I'm running Linux 6.1.63-current-meson64. I fixed the issue with the driver. The make script did not copy the new driver in the driver's folder. This was the error...

And I think that your problem is because you are using pymodbus-3.6.2.   Try with an old version: 2.5.3...
`sudo pip3 install -U pymodbus==2.5.3`

I read somewhere (I don't remember on) an issue with new version of pymodbus...

bistory, 2024-01-03:
> Well, I'm running Linux 6.1.63-current-meson64. I fixed the issue with the driver. The make script did not copy the new driver in the driver's folder. This was the error...
>
> And I think that your [...]

Thanks but @kasbert told me in an older issue (#57) that installing the driver is not needed anymore.
I will try to downgrade pymodbus to 3.2 as it seems to be working for him : https://github.com/kasbert/epsolar-tracer/commit/e154d96df5bac89fae6a92a63585b84e73163c86

bistory, 2024-04-15:
I upgraded recently to Debian Bookworm, installed Linux 6.6 and updated my scripts (adding rx and tx levels, brought by this PR : https://github.com/kasbert/epsolar-tracer/commit/7e30e9398683ae79ec495641ccc5744e42180be5 )
Everything is now working flawlessly !

closed 2024-04-15 by bistory