kasbert/epsolar-tracer #60: Error when trying to build the driver.

mahansford, 2024-03-14:
Was woking and installed but since a recent apt upgrade. 

when trying to build via DKMS I get this error:

`DKMS make.log for xr_usb_serial_common-1a for kernel 6.6.20+rpt-rpi-v8 (aarch64)
Thu 14 Mar 13:26:44 GMT 2024
make: Entering directory '/var/lib/dkms/xr_usb_serial_common/1a/build'
make -C /lib/modules/6.6.20+rpt-rpi-v8/build M=/var/lib/dkms/xr_usb_serial_common/1a/build
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
  CC [M]  /var/lib/dkms/xr_usb_serial_common/1a/build/xr_usb_serial_common.o
/var/lib/dkms/xr_usb_serial_common/1a/build/xr_usb_serial_common.c:1711:33: error: initialization of ‘ssize_t (*)(struct tty_struct *, const u8 *, size_t)’ {aka ‘long int (*)(struct tty_struct *, const unsigned char *, long unsigned int)’} from incompatible pointer type ‘int (*)(struct tty_struct *, const unsigned char *, int)’ [-Werror=incompatible-pointer-types]
 1711 |         .write =                xr_usb_serial_tty_write,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/xr_usb_serial_common/1a/build/xr_usb_serial_common.c:1711:33: note: (near initialization for ‘xr_usb_serial_ops.write’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /var/lib/dkms/xr_usb_serial_common/1a/build/xr_usb_serial_common.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /var/lib/dkms/xr_usb_serial_common/1a/build] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make: *** [Makefile:9: all] Error 2
make: Leaving directory '/var/lib/dkms/xr_usb_serial_common/1a/build'`

any ideas how I could resolve this and get it working :)

kasbert, 2024-03-17:
Please try with the standard kernel driver. The device is like /dev/ttyUSB0 .

ruggi99, 2024-05-03:
Hi @kasbert,
I removed the blacklist on the cdc-acm driver, updated initramfs and did a reboot of the Rapsberry.
After that I was having the /dev/ttyACM[0-1] devices (no /dev/ttyUSB0 device) but no one was working as I never received a response from the device.
So I compiled your driver thanks to the PR #61 (with the cdc-acm again in blacklist) and now it's working again and getting infos from the Epever.

```sh
pi@raspberrypi2:~ $ uname -r
6.6.28+rpt-rpi-v8
```

So from what I can see you expect that with kernel >= 6.6 the Epever device should work with the native kernel driver?

Are there any infos you want more?

Thanks

kasbert, 2024-05-05:
There should be /dev/ttyUSB0 device. I'l have to update and fire up my RPi some day,

mitchross, 2024-05-28:
> There should be /dev/ttyUSB0 device. I'l have to update and fire up my RPi some day,

please do, im in same boat!

kasbert, 2025-01-13:
I have now Raspbian 12/bookworm with kernel 6.6.62+rpt-rpi-v8 and it works with that kernel.

ruggi99, 2025-01-13:
I confirm it works out of the box now.
Thanks for keeping the driver up until now

closed 2025-01-15 by kasbert