kasbert/epsolar-tracer #64: More of a query

jimlynnjulian, 2025-02-06:
Just updated to Bookworm opsys on my RPi4b.
The driver is there, but doesn't seem to work.
Any clues, any reports?
Thanks.

`pi@raspberrypi:/var/www/html $ php client2.php
PHP Warning:  Specified serial port is not valid in /var/www/html/PhpSerial.php on line 120
PHP Warning:  Unable to set the baud rate : the device is either not set or opened in /var/www/html/PhpSerial.php on line 220
PHP Warning:  Unable to set parity : the device is either not set or opened in /var/www/html/PhpSerial.php on line 286
PHP Warning:  Unable to set length of a character : the device is either not set or opened in /var/www/html/PhpSerial.php on line 341
PHP Warning:  Unable to set the length of a stop bit : the device is either not set or opened in /var/www/html/PhpSerial.php on line 394
PHP Warning:  Unable to set flow control mode : the device is either not set or opened in /var/www/html/PhpSerial.php on line 456
PHP Warning:  The device must be set before to be open in /var/www/html/PhpSerial.php on line 146
PHP Warning:  Device must be opened in /var/www/html/PhpSerial.php on line 661
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576

`

kasbert, 2025-02-07:
Is /dev/ttyUSB0 present and is the client.php using it ?

jimlynnjulian, 2025-02-08:
`
The device doesn't seem to open. This is the data I've collected.

------------------------------------------------

ls /dev/ttyUSB*:
/dev/ttyUSB0

------------------------------------------------
client2.php

<?php
require_once 'PhpEpsolarTracer.php';
$tracer = new PhpEpsolarTracer('/dev/ttyUSB0');
$x=1;
while($x=1){
if ($tracer->getRealtimeData()) {
#	print "\nRealTime Data\n";
#	print "----------------------------------\n";
	for ($i = 0; $i < count($tracer->realtimeData); $i++)
#		print str_pad($i, 2, '0', STR_PAD_LEFT)." ".$tracer->realtimeKey[$i].": ".$tracer->realtimeData[$i].$tracer->realtimeSym[$i]."\n";
    print $tracer->realtimeData[$i].$tracer->realtimeSym[$i]."\t";
	} else print "Cannot get RealTime Data\n";
 print "\n";
$nano = time_nanosleep(60, 100000);
}
?>

------------------------------------------------

$>php client2.php
PHP Warning:  Specified serial port is not valid in /var/www/html/PhpSerial.php on line 120
PHP Warning:  Unable to set the baud rate : the device is either not set or opened in /var/www/html/PhpSerial.php on line 220
PHP Warning:  Unable to set parity : the device is either not set or opened in /var/www/html/PhpSerial.php on line 286
PHP Warning:  Unable to set length of a character : the device is either not set or opened in /var/www/html/PhpSerial.php on line 341
PHP Warning:  Unable to set the length of a stop bit : the device is either not set or opened in /var/www/html/PhpSerial.php on line 394
PHP Warning:  Unable to set flow control mode : the device is either not set or opened in /var/www/html/PhpSerial.php on line 456
PHP Warning:  The device must be set before to be open in /var/www/html/PhpSerial.php on line 146
PHP Warning:  Device must be opened in /var/www/html/PhpSerial.php on line 661
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
PHP Warning:  Device must be opened to read it in /var/www/html/PhpSerial.php on line 576
etc.
Cannot get RealTime Data
(usually occurs when access is too fast or some other app/service is blocking)

`

kasbert, 2025-02-08:
Do you have permissions to the /dev/ttyUSB0 ?
Try to run the program as root. If it works, grant access to your user. Do something like
https://forums.raspberrypi.com/viewtopic.php?t=259764

You should not need separate Maxlinear driver.

jimlynnjulian, 2025-02-08:
My results seem very similar yto yours:

Do you have permissions to the /dev/ttyUSB0 ?
Try to run the program as root. If it works, grant access to your user. Do something like
https://forums.raspberrypi.com/viewtopic.php?t=259764
Did not work with 'sudo su.'  <------------

Run ls -l and id commands. If the group is dialout, add that group to your user.

$ ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 Feb  8 11:56 /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Feb  8 13:04 /dev/ttyUSB0 <------------

$ id
uid=1000(kasper) gid=1000(kasper) groups=1000(kasper),
4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),104(input),106(render),108(netdev),997(gpio),998(i2c),999(spi)

uid=1000(pi) gid=1000(pi) groups=1000(pi),
4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),110(netdev),993(gpio),994(i2c),995(spi) <------------

kasbert, 2025-02-09:
I suppose you are using this [toggio/PhpEpsolarTracer](https://github.com/toggio/PhpEpsolarTracer.git)
I don't know why it is not able to open the device (error: ` Specified serial port is not valid in /var/www/html/PhpSerial.php on line 120`). 
Add some debug printing to PhpSerial.php line 88 and display values of `$this->_os` and `$device`

The client must configure the serial device. In python this is done by `serialclient.socket.rs485_mode = serial.rs485.RS485Settings(rts_level_for_rx=True,rts_level_for_tx=False)`
In PHP there is no such option. You have to configure the device with a separate program.

Clone and compile this repo: [rs485](https://github.com/kasbert/rs485)
```
$ ./rs485 /dev/ttyUSB0 send ''
```
This sets RS485 flags. I don't think this helps to the open error, but you have to do this anyway.

jimlynnjulian, 2025-02-09:
I hate when this happens. I compiled your repo and it works. While mulling over what to do next, I tried a compile of MAxlinear drivers. I tried others as well. This time on a fresh install of Bookworm 12. Same results nothing.
Then I created a python venv for the Epevermodbus software and ran a command line from the web:
epevermodbus --portname /dev/ttyUSB0 --slaveaddress 1glad.
Very much to my surprise, the program ran and I started getting data. I tried an accompanying script and that ran.
I suspect when I install the PHP software and PHP, that too, will run without incident.
I'm just not sure what I did or if the command line triggered a change in settings. 
I don't know, but I'm glad.
Thanks for all you assistance. You've been a great help.

jimlynnjulian, 2025-02-09:
I may have goten the thing working by accident.
./rs485 /dev/ttyUSB0 receive "$3100". Honestly, I'm not sure what that did. I know the register is one of the read registers in the Epever charge controller. I'm not sure how the send command triggered the driver, though. I need to make this permanent, The access disappears when rebooting.

over_voltage_disconnect_voltage: 14.8
charging_limit_voltage: 14.6
over_voltage_reconnect_voltage: 14.2
equalize_charging_voltage: 14.6
boost_charging_voltage: 14.6
float_charging_voltage: 14.4
boost_reconnect_charging_voltage: 14.2
low_voltage_reconnect_voltage: 12.4
under_voltage_recover_voltage: 12.0
under_voltage_warning_voltage: 11.6
low_voltage_disconnect_voltage: 11.2
discharging_limit_voltage: 11.2
boost_duration: 180
equalization_duration: 0

closed 2025-02-25 by jimlynnjulian